jaredjennings/dactyl-keyboard #11: run scad-clj changes upstream

jaredjennings, 2018-05-10:
Interesting features have been added to scad-clj to enable the use-openscad-modules branch. In order to run with the modified scad-clj, I've added checkouts/scad-clj as a symlink to my scad-clj working copy. It'd be much nicer to be able to just write in the project.clj, "we need version 3000 of scad-clj," rather than continue this rigmarole and have to guide new users in it.

jaredjennings, 2018-05-27:
Filed farrellm/scad-clj#2

farrellm, 2018-07-09:
I just pushed version 0.5.3 of scad-clj to clojars with all the recent changes, including farrellm/scad-clj#36. 

In the future, please feel free to reach out if you need me to tag a new jar. I get busy and/or lazy, so I often wait for a few commits to tag. But if there is urgency, I'm happy to update.

jaredjennings, 2018-07-11:
Wonderful! Thanks, @farrellm!

closed 2018-07-11 by jaredjennings