jaredjennings/dactyl-keyboard #46: Combining with RollerMouse Red

dfenn, 2018-09-03:
I'm interested in how you've split this up into pieces. I have a 3D printer with a small printing volume, so the only way I'd be able to print it would be in pieces.

I'd like to combine a Dactyl with the RollerMouse Red (https://www.contourdesign.com/product/rollermouse-red/). To do that, I think the roller would have to butt pretty close up against the thumb cluster, so that the keyboard wouldn't be too far away. 

I think that to do that, it would be necessary to remove the marshmallow part on the front edge, and the height may have to be adjusted. I've looked through the source files, but I'm not sure where to start to accomplish these.

Can you give me any pointers about how I might accomplish that?

Thanks!

jaredjennings, 2018-09-04:
@dfenn, welcome! The height is adjusted in things/key-place.scad in the KeyPlace and ThumbPlace modules. Since it's raw OpenSCAD code, if you adjust it while you have dactyl-top-right-all.scad open in OpenSCAD, you'll see the changes immediately.

For the marshmallow part, first you would need to make the sides not go all the way around the keyboard, and then you'd have to figure out how to split the resultant sides in pieces and how to attach them.

The sides are defined in sides.clj; right at the end, [`sides-right`](https://github.com/jaredjennings/dactyl-keyboard/blob/11f57b17ae8c23b1b7689550111812c5fe6c5862/src/dactyl_keyboard/sides.clj#L352) is defined as a closed sides object (last argument true) whose, ah, shall we say points, are `around-edge`. You probably want to make a copy of [`:around-edge`](https://github.com/jaredjennings/dactyl-keyboard/blob/11f57b17ae8c23b1b7689550111812c5fe6c5862/src/dactyl_keyboard/layout.clj#L38), which is in layout.clj's `:y-and-b` layout, and call it `:sides-exist-at` or something. (`layouts` was supposed to be a way to define multiple sibling keyboards, e.g., different sizes, and choose one to render. That has not happened yet. See #37 and #38.) [Below the `def layouts`](https://github.com/jaredjennings/dactyl-keyboard/blob/11f57b17ae8c23b1b7689550111812c5fe6c5862/src/dactyl_keyboard/layout.clj#L125), around line 100, are a bunch of mentions of `chosen-layout`. That's how the key `:around-edge` in the `:y-and-b` layout becomes the global variable `around-edge`. So add a new def there for `sides-exist-at`. Now make `sides-right` at the end of sides.clj use `sides-exist-at` instead of `around-edge`, and pass false as the last parameter because your sides shape is no longer closed.

To change how the sides are split into pieces, change [`:sides-partitions`](https://github.com/jaredjennings/dactyl-keyboard/blob/11f57b17ae8c23b1b7689550111812c5fe6c5862/src/dactyl_keyboard/layout.clj#L44) in layout.clj (see output file `splits-right.scad` for the results of your changes), and change `:sides-slice-joints`, immediately following, to match.

The sides are glued to the frame using the [`:sides-frame-joints`](https://github.com/jaredjennings/dactyl-keyboard/blob/11f57b17ae8c23b1b7689550111812c5fe6c5862/src/dactyl_keyboard/layout.clj#L51). Some of these will no longer join the frame to anything once you have removed the front of the thumb, so you'll have to change them.

All of these things are specified using a family of rather compact notations. The somewhat exact meaning of the `:sides-slice-joints` notation, for one, is [in the docstring of `intersects-from-notation` in sides_pieces.clj](https://github.com/jaredjennings/dactyl-keyboard/blob/11f57b17ae8c23b1b7689550111812c5fe6c5862/src/dactyl_keyboard/sides_pieces.clj#L32). `:around-edges` notation eventually gets passed to [`sides-place-fns`](https://github.com/jaredjennings/dactyl-keyboard/blob/11f57b17ae8c23b1b7689550111812c5fe6c5862/src/dactyl_keyboard/layout.clj#L221), and is documented there.

Generally, though, `k` means key-place, and `t` means thumb-place. On the right side of the keyboard, which is the primary one modelled, `k` column numbers (-1 through 5 inclusive, see `:columns-pieces`) go up as you go to the right (aka east), and row numbers (0 through 4 inclusive, see `:rows`) go up as you move away from the user (aka north). `t` column numbers (0 through 2 inclusive) go up as you go to the left, and row numbers (-1 through 1 inclusive) go up as you move away from the user. The details, like `[:n :k 3 2]` or `[:w-of-k 3 2]`, are modifications of those places on the keyboard, inexpertly specified.

Hope this helps; please ask more questions.

(You know, looking at it from the top, you might want to shave some of the marshmallow off of the near right corner too... hmm.)

dfenn, 2018-09-05:
@jaredjennings, thank you for the detailed response. I'm hoping to have some time this weekend to look into this. I'll keep you posted on how it goes, and I'll ask you more questions as necessary.

Thanks!

jaredjennings, 2018-09-09:
I've done the first part of this in 62da0d1ceb3b29cdbd01853ff356c28cdd277459 (no attention to how the sides are split into pieces, nor attached to the frame). This brings up the issue that the bottom is still defined under a place where the sides don't exist -- screenshot: https://imgur.com/Ksts5kO.

jaredjennings, 2018-09-18:
A possibly simpler thing to do is to change the shape used to form the marshmallow. Twice in sides.clj, the form ```(sphere r)``` occurs. If you change that to, say, ```(scale [0.5 0.5 1] (sphere r))```, it doesn't stick out as far in front of the keyboard. (Nor as far out from any of the sides.) Any convex shape will do - I've tried it with cubes and cylinders of various rotations, too. The reason it has to be convex is that a piece of it is hulled with a connector body to make the sides-frame connector, so if it's concave, you'll have bumps on the outside wherever there's a connector.

You know, it might be fun to vary the shapes hulled by position. If you were to use a cylinder rotated onto its side for the southernmost positions, you could achieve a flat front while still having the case exist there, and it would probably blend well with the spheres used everywhere else. (To be clear, the code as it stands doesn't support that. But it would be a pretty small change.)

dfenn, 2018-10-01:
Sorry for the novice question here, but how do you view the side pieces? When I view the sides-right-0*.scad files, not much shows up. It just looks like some very small pieces. 

Also, are the sides required for structural support, or are they purely aesthetic? If they're just aesthetic, I may try doing a preliminary print without them just to see how the keyboard would feel with the RollerMouse. I'd rather not invest a lot of time tweaking things if it turns out that the Dactyl and the RollerMouse are going to be a bad combination anyway.

dfenn, 2018-10-02:
Turns out that was user error on my part. This is the first time I've used OpenSCAD, and I didn't realize there was a difference between previewing and rendering. The previews weren't working right, but as soon as I rendered, I was able to see everything fine.

jaredjennings, 2019-12-29:
Closing due to inactivity.

closed 2019-12-29 by jaredjennings

jaredjennings, 2021-11-05:
https://www.reddit.com/r/ErgoMechKeyboards/comments/qmujxm/lily58_wireless_with_rollermouse_setup/