jaredjennings/dactyl-keyboard #59: Automatic placement of the switches footprints on a pcb

Supermagnum, 2020-08-17:
Since the position of the holes for the switches is known, there should be possible to add a utility that places the footprints of the switches on a PCB according to the geometric layout of the keyboard halves.
Placement on the pins for ALPS switches:
https://matias.ca/switches/click/blueprints/3.jpg
Cherry switches:
https://www.cherrymx.de/en/dev.html

Hybrid footprints for Kicad ( he should have used a oval hole, its easy to spot where )
https://github.com/ai03-2725/MX_Alps_Hybrid

Kicad has a python API:
https://docs.kicad-pcb.org/master/en/plugins/plugins.html

jaredjennings, 2020-08-17:
This is a great idea. It requires flexible printed circuits (FPC), or [wiggles and weaves](https://www.reddit.com/r/ErgoMechKeyboards/comments/i83bw5/wiring_ideas_for_the_dactyl_wiggles_and_weaves/), owing to the 3D nature of this keyboard. It's not a high priority for me, because so far I've just handwired them. I'll also note that gEDA PCB has a text-based file format. I'm not sure how easy KiCAD's file format is to write, or how stably it is defined, but this might be an easier way to integrate with it than talking to a running instance of KiCAD. Although KiCAD is more commonly used, I haven't personally found anything to dislodge me from PCB yet.

Patches gratefully accepted!

jaredjennings, 2020-08-31:
Maybe it only requires [0.6mm PCBs](https://www.reddit.com/r/MechanicalKeyboards/comments/ijwoe9/made_a_flexible_pcb_for_dactyl_dactyl_manuform/)? - This implementation of PCBs is not parameterized, and the keyboard was remodelled in Fusion 360, with the key positions fixed. So automatic PCB generation would still be novel.

This feature would likely require the key placement function to be usable in Clojure, not only OpenSCAD (#10): OpenSCAD can project 3D features down to 2D, but that is different from unfolding them to a flat plane. And now that I describe this that way, it sounds a lot like the problem of making paper mockups, for which I can't find an issue. Could have sworn I filed that one.

jaredjennings, 2020-08-31:
Well, that paper mockup ticket is now #61.

jaredjennings, 2020-09-07:
More on that one with the 0.6mm PCB, the [Bastyl](https://www.reddit.com/r/ErgoMechKeyboards/comments/inozxt/bastyl_with_flexible_pcb_mcu_holder_flush_usb_port/)

Supermagnum, 2020-09-07:
The PCB'S should optimally have one or two signal layers and two other layers acting as shield to minimize the risk for radio interference.

jaredjennings, 2020-09-08:
If the code to generate the PCB were written, adding the two extra layers would be a small extra effort. But I don't think it's necessary: the original Dactyl, and the Kinesis Advantage, are made with single-layer flexible printed circuits (FPCs). The signals travelling over these boards are maximally in the kilohertz range - all the multi-MHz signals where this consideration comes into play the most are on the Pro Micro or Teensy itself.

Supermagnum, 2020-09-09:
This is a better footprints library, it is scheduled to be merged in to the official Kicad GitHub library:
https://github.com/egladman/keebs.pretty
Merge request:
https://github.com/egladman/keebs.pretty/issues/7

jaredjennings, 2020-10-19:
The [Bastyl](https://www.reddit.com/r/ErgoMechKeyboards/comments/jdyc1q/bastyl_ergonomical_split_keyboard_3d_printed_case/) folks have managed this without wiggles and weaves, and without FPCs. (Good discussion about those in the comments, particularly cost/build-complexity tradeoffs.)

jaredjennings, 2020-10-19:
[Discussion](https://www.reddit.com/r/ErgoMechKeyboards/comments/jd2t6z/prototype_dactylmanuform_pcb/) about commercial use. It seems that the friend commercializing flexing FR4 PCBs for ergonomic keyboards on [cy384's blog](http://www.cy384.com/blog/flex-pcbs.html) ("wiggles and weaves" above) may have been u/smyjpmu at hidtech.ca?

This is something to watch out for: HIDTech's design appears to have been licensed CC-BY-NC, which is more restrictive than this project. So any appearance of copying that design here should be avoided. In fact, it may be impossible to develop this feature unimpeachably at this time.

jaredjennings, 2021-02-05:
More about HIDTech, Bastyl, u/smyjpmu (Anthony), and u/fmcraft (Quentin): https://www.reddit.com/r/ErgoMechKeyboards/comments/ld97a3/psa_anthony_usmyjpmu_is_a_con_artist_who_scammed/

equalunique, 2021-02-06:
A similar utility was created by Phil Hagelberg for his Atreus project: https://gitlab.com/technomancy/atreus/-/blob/efcf961441966f0dfb305b80bf78b3da62ca0f3f/atreus.rkt

Supermagnum, 2021-04-23:
Any updates on this?

jaredjennings, 2021-04-25:
I haven't done anything on the topic. Quentin's got PC boards at https://bastardkb.com, but they won't fit this keyboard with all of my layout tweaks. Thanks to a working prototype I made in the past few weeks, I've found my tweaks are horrible anyway, so go buy one of his keyboards :)

jaredjennings, 2021-08-23:
https://www.reddit.com/r/ErgoMechKeyboards/comments/p8m8h0/dactyl_flex_pcbs_arrived_from_jlcpcb_with/