jaredjennings/dactyl-keyboard #60: Support for raw data from http://www.keyboard-layout-editor.com

Supermagnum, 2020-08-17:
I think that support for raw data from the mentioned page should be implemented.
I dont understand Clojure at all.

A example :
http://www.keyboard-layout-editor.com/#/gists/c881b4af35745d286a7193ede6e3d1f9

jaredjennings, 2020-08-17:
This is a fantastic idea. The 2D coordinates from KLE could not be used directly, but the information they convey about adjacency and size of keys would be quite useful.

With that said: I spent a long period of time trying to make the code more flexible, so that I could describe the keyboard I wanted at a layer above the code. The more I tried to do that, the richer my keyboard description needed to become. After a while, I got tired of trying to build a keyboard language and not having a keyboard of my own, and decided to describe the keyboard using Clojure code. And I'm still not done with that!

So I do have a long-term goal of making it easier for people to prototype, build, and program ergonomic keyboards, and taking data from KLE would be an important step toward that. But before I get there, you could probably learn Clojure and write it up yourself.

Supermagnum, 2020-08-17:
I don't think I am capable of doing that.

Supermagnum, 2020-08-18:
Reddit user : https://www.reddit.com/user/ChirsF/ writes
Get a regex101 with the format output and separately what the expected input is and I’ll take a crack at a regex for it.
 in this thread:
https://www.reddit.com/r/ErgoMechKeyboards/comments/ibm4y0/feature_request_for_all_daktyl_keyboard_github/?utm_medium=android_app&utm_source=share

jaredjennings, 2020-08-18:
It's just JSON, isn't it?

Supermagnum, 2020-08-20:
> It's just JSON, isn't it?

Looks like it is.